joelsernamoreno/EvilCrow-RF #2: Schematics

leo-lb, 2021-05-02:
Hello!

Wonderful cheap device! As soon as I can get one I am looking forward to making it easier to use by creating better user interfaces and so on.

In the mean time, I would be curious to look at the schematics, where can I find them?

Thank you

joelsernamoreno, 2021-05-03:
Check your email :)

closed 2021-05-03 by joelsernamoreno

NicheSkill, 2023-10-02:
@joelsernamoreno Would you please email me the schematics as well ?
I have managed to get the device working till the home page load in browser. however I want to know  hardware schematics and connections for further modules. 
basically this borad is not available in india so trying to build my own. I got cc1101 and NRF24L modules  but not sure how can i interface it.