joelsernamoreno/EvilCrow-RF #3: Steady RF Carrier output?

juergs, 2021-05-05:
[image]

Experiencing a steady RF Carrier, when pluged into USB?

Which functionality have LED 1..4?

Schematic?

joelsernamoreno, 2021-05-05:
Hi!

That's strange. Check my image:

[image]

Can you test this by connecting the device to an external battery? It is strange that Evil Crow RF generates that carrier.

About the leds... you can turn off/on them using these pins:

* led_1 = io12
* led_2 = io13
* led_3 = io15

About the schematics, tell me your email and I will send you 

Thanks!

joelsernamoreno, 2021-05-05:
Can you check this with this code?

#include <ELECHOUSE_CC1101_SRC_DRV.h>
#include <RCSwitch.h>

RCSwitch mySwitch = RCSwitch();
RCSwitch CC1101_1 = RCSwitch();
RCSwitch CC1101_2 = RCSwitch();

//CC1101 Pins
int sck = 18;
int miso = 19;
int mosi = 23;
int csn1 = 5;
int csn2 = 27;
int gdo0_1 = 2;
int gdo2_1 = 4;
int gdo0_2 = 25;
int gdo2_2 = 26;

void setup() {
Serial.begin(38400);
ELECHOUSE_cc1101.setSpiPin(sck, miso, mosi, csn1);
ELECHOUSE_cc1101.setMHZ(433.92);
ELECHOUSE_cc1101.Init();
CC1101_1.enableTransmit(gdo0_1);
ELECHOUSE_cc1101.SetTx();
CC1101_1.send(1648498, 24);
delay(3000);
CC1101_1.send(1648498, 24);
delay(3000);
ELECHOUSE_cc1101.SetRx();
}

void loop() {
}

juergs, 2021-05-06:
Hi Joel,
thanks for that quick response. ;-)

I had overwritten the original flash-content and used your github-code, without modifications on the Elechouse-Lib.

I'll  check with your suggested testcode. 

Thank you!
Juergen

juergs, 2021-05-06:
Hi Joel,

 

thank you very much for your work & time.

 

My email-address:          ***@***.*** ***@***.***> 

 

I would appreciate to get a schematic, to check the error. 😉

Might be,  that I haven’t  used the second Radio, by setting it to receive mode?

 

We‘ll try to use this device for our german Homeautomation-System FHEM (https://fhem.de/)  as RF-gateway-device.

 

thank you + regards,

Juergen

joelsernamoreno, 2021-05-06:
Hello!

This has nothing to do with setting up the second radio. You can use one radio or use both radios. For both radios to work correctly you need to use the libraries included in the libraries folder. These libraries have been modified to make the device work correctly.

Try the example I have sent you. Possibly the carrier you see is a problem of the basic firmware. I have talked to some colleagues who have Evil Crow RF and there is one friend who has this problem too. 

If the example I sent you works and you don't get the carrier, I will tell you how to fix this in the basic firmware

I look forward to your reply :)

Thanks!

juergs, 2021-05-06:
[image]

Other, but similar behavior: Carrier is there, but interrupted.

joelsernamoreno, 2021-05-06:
Can you try transmitting with the second module?

juergs, 2021-05-06:
Yes. Receiving with original FW was not possible ... with obvious "powerful" traffic nearby ... (;-(

juergs, 2021-05-06:
[image]

_ELECHOUSE_cc1101.setSpiPin(sck, miso, mosi, csn1);
ELECHOUSE_cc1101.setMHZ(433.92);
ELECHOUSE_cc1101.Init();

CC1101_1.enableTransmit(gdo0_1);
ELECHOUSE_cc1101.SetTx();

CC1101_1.send(1648498, 24);
delay(3000);
CC1101_1.send(1648498, 24);
delay(3000);
ELECHOUSE_cc1101.SetRx();

delay(3000);

ELECHOUSE_cc1101.setSpiPin(sck, miso, mosi, csn2);
ELECHOUSE_cc1101.setMHZ(433.92);
ELECHOUSE_cc1101.Init();

CC1101_2.enableTransmit(gdo0_2);
ELECHOUSE_cc1101.SetTx();

CC1101_2.send(1648498, 24);
delay(2000);
CC1101_2.send(1648498, 24);
delay(2000);
ELECHOUSE_cc1101.SetRx();_

joelsernamoreno, 2021-05-06:
Can't receive with the second module? For the second module to work you need to copy the libraries from the libraries folder (rcswitch and SmartRC) to your arduino libraries directory

About the schematics, check this: https://github.com/joelsernamoreno/EvilCrow-RF/tree/main/schematics

Can you try transmitting with the rfquack firmware and see if you get the carrier?

juergs, 2021-05-06:
Thank you, i will check, but will take some time ...

joelsernamoreno, 2021-05-06:
Okay, let me know if you have any problems with RFQuack

It is strange that you see that carrier, this is why I would like you to try the RFQuack firmware. I want to know if this is a firmware problem or a hardware problem

If it is a hardware problem you will have to talk to the manufacturer to send you a new one. Right now they are out of stock, but hopefully they will have stock soon

juergs, 2021-05-06:
https://e2e.ti.com/support/wireless-connectivity/sub-1-ghz/f/sub-1-ghz-forum/872424/cc1101-how-to-send-unmodulated-carrier-using-cc1101

compiling .pio\build\rf\libbe6\SmartRC-CC1101-Driver-Lib\ELECHOUSE_CC1101_SRC_DRV.cpp.o
libraries\SmartRC-CC1101-Driver-Lib\ELECHOUSE_CC1101_SRC_DRV.cpp: In member function 'void ELECHOUSE_CC1101::setPA(int)':
libraries\SmartRC-CC1101-Driver-Lib\ELECHOUSE_CC1101_SRC_DRV.cpp:403:16: warning: 'a' may be used uninitialized in this function [-Wmaybe-uninitialized]
 PA_TABLE[0] = a;
                ^

joelsernamoreno, 2021-05-06:
What do you want to tell me with this?

joelsernamoreno, 2021-05-06:
Anyway, can you tell me what SDR hardware you are using and what software? I will try to replicate your scenario

Example: hackrf with gqrx

juergs, 2021-05-06:
SDRSharp community package (Win10): [(https://airspy.com/download/)]
and USB-DVBT-Stick.

May be the antennas position is to near to transceiver ? 
2m away there are some carrier traces visible.
[image]

joelsernamoreno, 2021-05-06:
That should not affect

I have spoken to LSatan, it seems to be a library problem. This is the response from LSatan:

"yes i know the narrow carrier signal. we should set set setSidle after sending. or SetRx. that should fix the problem."

Try the rfquack firmware and tell me if the carrier is still there

juergs, 2021-05-06:
Try the rfquack firmware and tell me if the carrier is still there.
I am searching all needed libs  ;-)

..\RFQuack\examples\RFQuack-esp32-CC1101-serial\src\main.cpp
..\RFQuack\src\radio

joelsernamoreno, 2021-05-06:
You can see the RFQuack section in my repository. In this section you have the steps to flash RFQuack in Evil Crow RF

https://github.com/joelsernamoreno/EvilCrow-RF/blob/main/README.md

juergs, 2021-05-06:
[image]

Seems to work now:

  ELECHOUSE_cc1101.setSpiPin(sck, miso, mosi, csn1);
  ELECHOUSE_cc1101.setMHZ(433.92);
//*************************************
  ELECHOUSE_cc1101.setPA(10);
//*************************************
  ELECHOUSE_cc1101.Init();

  CC1101_1.enableTransmit(gdo0_1);
  ELECHOUSE_cc1101.SetTx();

  CC1101_1.send(1648498, 24);
  Serial.println("TX_1)"); 
  delay(3000);

  CC1101_1.send(1648498, 24);
  Serial.println("TX_2)"); 
  delay(3000);
  
  CC1101_1.send(1648498, 24);
  Serial.println("TX_3)"); 
  delay(3000);

  CC1101_1.send(1648498, 24);
  Serial.println("TX_4)"); 
  delay(3000);

  ELECHOUSE_cc1101.SetRx();
  delay(3000);

//---------------------------------------------------
... and set PKTLEN = 0xFF

   //SpiWriteReg(CC1101_PKTLEN,   0x3D);
    SpiWriteReg(CC1101_PKTLEN,   0xFF);

joelsernamoreno, 2021-05-06:
Well, I'll modify this in the repository library

The SmartRC library in my repository is version v2.3. I am using this version because I had to modify the library to make both CC1101 modules work at the same time. I have also made this modification in the latest version of SmartRC, but in the latest version the two radio modules do not work at the same time even though I modified the library.

When I have free time I will modify the spi register with your change. I think setPA(10) has nothing to do with that carrier. SetPA is for increasing the transmit power. Anyway, if you want to use the basic firmware from my repository you can add setPA in the code. 

LSatan and I are trying to make the latest version of SmartRC compatible with Evil Crow RF, but we haven't managed to make both CC1101 modules work at the same time yet. If you want you can help on this

Let me know if I can close this issue. Thank you very much!

juergs, 2021-05-06:
Thank you for your kind support!

RX is working, too     :-)

PartNum: 0
VERSION: 20
Module: 1
Frequency: 433.92
Bandwidth: 58.00
Received: 1648496 BIN: 110010010011101110000 / Codelen:24bitsProtocol: 5
Received: 1648496 BIN: 110010010011101110000 / Codelen:24bitsProtocol: 5
Received: 1648496 BIN: 110010010011101110000 / Codelen:24bitsProtocol: 5
Received: 1648498 BIN: 110010010011101110010 / Codelen:24bitsProtocol: 5

juergs, 2021-05-06:
You can close this issue. 

>CC1101 modules work at the same time yet. If you want you can help on this
We have a 4 CC1101-module transceiver with Maplemini STM32F103 solution:
Take a look @:
[(https://github.com/Ralf9/SIGNALDuino/blob/dev-r412_cc1101/SIGNALDuino.ino) ] 
for FHEM Homeautomation system.

closed 2021-05-06 by joelsernamoreno