joelsernamoreno/EvilCrow-RF #4: Add new features.

djecom1, 2021-05-07:
It's not realy a issues, but maybe someone can help me to add new features on the original firmware.
I found this repo (link below) and tried adding a "De Bruijn" part, but no luck so far.
I will try again, but maybe someone here could find an easier way than me! 

[gusgorman/RFmoggy](https://github.com/gusgorman402/RFmoggy)

And many thanks Joel for this nice tool.

Jérôme.

joelsernamoreno, 2021-05-07:
Hello!

I can try to implement this with the basic firmware, but... have you tried this with Evil Crow RF and does it work?

This code uses ESP8266, possibly you have to change some things to make it work on ESP32 :)

joelsernamoreno, 2021-05-09:
I have adapted De Bruijn to work with Evil Crow RF. You can download this here:

https://mega.nz/file/x0ZT1QqL#RfCf--2aYiGoUCGTm1-YWoMBQLRcGm4GIEeIIM-_VO8

Try this and tell me if it works correctly

I have tested this but it seems strange to me. Bruijin transmits only once and gets stuck

djecom1, 2021-05-09:
At work today, I'll try it tomorrow.
Thank you!

joelsernamoreno, 2021-05-10:
Yes, you have to make some modifications to make it compile

I have adapted RFMoggly completely for Evil Crow RF. This afternoon I will upload the code to the repository. Be patient :)

joelsernamoreno, 2021-05-10:
The web application does not break when you run anything?

For example: When after transmitting with De bruijn you have to receive a message saying: signal sent. If you don't get this message it is because the code has broken

djecom1, 2021-05-10:
No problem :
"Signal is sent
 Return to main menu
 Return to XXX menu" 
👍

closed 2021-05-10 by joelsernamoreno