joelsernamoreno/EvilCrow-RF #5: RollJam attack

sumanblack666, 2021-05-09:
Hi ..bought this at aliexpress and still waiting to receive it ...cant wait to play with it ...wondering why there's no samy's rolljam attack module available , i saw its available at https://github.com/rfquack/RFQuack/tree/8a9364766192ae660c95ee0392152190ab1d1176/src/modules/defaults ?

is this due to security reasons ?

joelsernamoreno, 2021-05-09:
Hi!

Are you talking about adding rolljam in the basic firmware?

sumanblack666, 2021-05-09:
Yes ..

joelsernamoreno, 2021-05-09:
I don't have that much free time to implement features that fast. We are currently working on a library for Evil Crow RF that does not depend on rc-switch. We're also working on properly adapting the latest version of SmartRC here. With the latest version of SmartRC there are some problems to use both radio modules at the same time, so the library in this repository is version 2.3.

The community can add new features to Evil Crow RF, you can PR if you have something new for the device :)

sumanblack666, 2021-05-09:
Understood .I'll try to implement once i receive the device .

closed 2021-05-09 by sumanblack666

joelsernamoreno, 2021-05-09:
Thank you!

I hope you understand this. The devices are developed in our free time and we don't make money with this, so we try to make them as cheap as possible

We can't offer a great feature support like hak5 or Great Scott Gadgets

LeonSmall512, 2023-08-01:
How much will u charge me to make a rolljam device.