joelsernamoreno/EvilCrow-RF #6: Receiver granularity

leo-lb, 2021-05-11:
Hello!

I am wondering how this device could be used as a standalone tool and will it be sufficient, in particular I am thinking of a spectrum analyzer, is it going to be precise enough or would you still need an RTL-SDR?

I am thinking of creating an Android app with easy autocompletion of RFQuack commands and some UI for specific tasks like signal analysis or spectrum analyzer.

Can you think of any limitations in the hardware that would prevent this from working?

Thank you

leo-lb, 2021-05-11:
> Hi!
> 
> What do you mean when you say a standalone tool?
> 
> About the limitations... this is not a SDR, there are many limitations because all the radio part is processed in the hardware

I mean that you wouldnt need any other hardware to study RF protocols on the supported bands and retransmit/emulate/eavesdrop those protocols, of course given the available bandwidth etc. of the chips.

joelsernamoreno, 2021-05-11:
Depending on the signal you are analyzing. If the signal has some kind of encryption, you may need an sdr with other tools

leo-lb, 2021-05-11:
@joelsernamoreno I read you saying elsewhere that it was possible to access raw I/Q, is it possible to receive and transmit raw I/Q? Or are we limited to encode/decode present in hardware?

leo-lb, 2021-05-11:
@joelsernamoreno Yes this, thank you ! So both sides is supported, so why exactly would signal encryption be a problem to break here with the hardware? Thanks

joelsernamoreno, 2021-05-12:
If you receive the raw and transmit the raw no problem, but the interesting thing is to analyze the signal and see what it contains :)

leo-lb, 2021-05-12:
@joelsernamoreno Sure but I mean that in any case there wont be any limitation on what can be received or transmitted (besides bandwidth), correct?

Also what prevents C1101 to be plugged with GNU Radio considering it has raw TX/RX IQ?

joelsernamoreno, 2021-05-12:
I think that CC1101 modules cannot be connected to gnuradio. As I have told you before, this is not a SDR like rtl-sdr, hackrf, etc

joelsernamoreno, 2021-05-12:
Evil Crow RF is not an SDR. It is simply a "breakout board" for the TI CC1101 platform. It is not as flexible as an SDR. It has a limited set of input parameters that you can define and it will decode the data for you

leo-lb, 2021-05-13:
> Evil Crow RF is not an SDR. It is simply a "breakout board" for the TI CC1101 platform. It is not as flexible as an SDR. It has a limited set of input parameters that you can define and it will decode the data for you

If you can recv/send raw I/Q then it doesnt decode data for you?

leo-lb, 2021-05-13:
> I think that CC1101 modules cannot be connected to gnuradio. As I have told you before, this is not a SDR like rtl-sdr, hackrf, etc

I am thinking it may be about writing the driver?

joelsernamoreno, 2021-05-13:
You can receive raw data and decode, but this does not mean you can use the device with gnuradio, urh, gqrx, etc

There are a cc1101 plugin for urh, but I have never tested this:

https://github.com/SpaceTeddy/urh-cc1101_FEC_encode_decode_plugin

joelsernamoreno, 2021-05-13:
Read this: https://www.barrettcommunications.com.au/news/understanding-the-difference-between-sdr-and-hdr/

leo-lb, 2021-05-13:
Thank you! Reading!

joelsernamoreno, 2021-05-13:
I am closing this issue. If you have any doubt open another issue :)

closed 2021-05-13 by joelsernamoreno