joelsernamoreno/EvilCrow-RF #8: next production batch ?

timFaivre, 2021-05-12:
Hi,
sorry in advance to reach you here, but, ... 

I see the hardware as unvailable on aliexpress : do you plan to lauch a second batch anytime soon ? I would be interested. Perhaps using croud supply if you need financing ?

joelsernamoreno, 2021-05-12:
Hello!

A few days ago we made the second batch with 29 devices. This sold out very fast

The manufacturer is making 100 devices to sell. I hope it doesn't take them too long to make this available for sale

I will write you here when Evil Crow RF is available :)

timFaivre, 2021-05-12:
Awesome ! I'm staying tuned.

MrTakfly, 2021-05-23:
It works every time in Windows 👍 Thank you for your help!

I'm currently implementing a multi-bank memory for recording and replaying signals. It's a bit clunky at the minute, but once I've polished it I will add a web interface and submit a PR! Again, great project! You're a wizard!

Pr0l1f1c, 2021-05-24:
Great tool, I'm excited myself to buy a few for my team! Thanks!

joelsernamoreno, 2021-05-24:
Hi
I think the stock will return this week

opless, 2021-05-24:
Hi, sorry for the comment spam, I'm also interested so commenting to receive updates.

gigawatts, 2021-06-01:
28 in stock right now

joelsernamoreno, 2021-06-01:
Stock!

Pr0l1f1c, 2021-06-01:
Thx for the heads up! I need a total of 28 for a class room but settled for 4 this time to allow others a chance to grab one. Thx again!

joelsernamoreno, 2021-06-01:
Hi @Pr0l1f1c 

Hello

You can write me on twitter (@JoelSernaMoreno) and I will put you in contact with the manufacturer if you need a batch of several devices

Thanks!

joelsernamoreno, 2021-06-01:
I close this issue. If you ever need to know when new stock is available, please open a new issue

Thanks

closed 2021-06-01 by joelsernamoreno