joelsernamoreno/EvilCrow-RF #9: Uploading sketches to the device

MrTakfly, 2021-05-23:
Hi, I seem to have very hit and miss results when uploading firmware to the device.

I followed the installation instructions and it failed to connect multiple times, but then pressing the reset button when the device was mid connecting seems to work around 1 in 20 times. Is there a trick to this that I have missed?

Awesome project by the way! I will be submitting a PR with some funky features soon.

joelsernamoreno, 2021-05-23:
Hi!

What operating system are you using?

There are some people who have had the same problem in Debian, in Windows or Ubuntu there should not be this problem

MrTakfly, 2021-05-23:
Thanks for the speedy reply!

I'm running Debian so that may be my issue! I'll try again on my windows box and see what happens! Thanks dude!

joelsernamoreno, 2021-05-23:
When I have some free time I will try this in debian. I have tried this on ubuntu and windows and I have no problem uploading the code

Let me know if this works fine on your windows

closed 2021-05-23 by joelsernamoreno

MrTakfly, 2021-05-23:
It works every time in Windows 👍 Thank you for your help!

I'm currently implementing a multi-bank memory for recording and replaying signals. It's a bit clunky at the minute, but once I've polished it I will add a web interface and submit a PR! Again, great project! You're a wizard!

joelsernamoreno, 2021-05-23:
Oh, that sounds good. Thanks for helping with the device! :D