joelsernamoreno/EvilCrow-RF #10: Putting unit in programming mode

RChadwick7, 2021-05-28:
Is there a certain button combination that puts the Evil Crow in programming mode? Every time I try to flash RFQuack, it fails.

joelsernamoreno, 2021-05-28:
Are you using debian? It seems that debian does not correctly detect the device 

In ubuntu it works perfectly

RChadwick7, 2021-05-29:
I'm using Ubuntu 16.04. This is the result of sudo PORT=/dev/ttyUSB0 make flash:

docker run --rm -it --device=/dev/ttyUSB0:/board --env-file build.env rfquack/rfquack
Installation ID is 19b4a875-dbab-4d58-a017-58220b7a2dde
Processing ESP32 (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)
Tool Manager: Installing platformio/framework-arduinoespressif32 @ ~3.10006.0
Downloading [####################################] 100%
Unpacking [####################################] 100%
Tool Manager: framework-arduinoespressif32 @ 3.10006.210326 has been installed!
Tool Manager: Installing platformio/tool-mkspiffs @ ~2.230.0
Tool Manager: tool-mkspiffs @ 2.230.0 has been installed!
Tool Manager: Installing platformio/tool-scons @ ~4.40100.2
Downloading [####################################] 100%
Tool Manager: tool-scons @ 4.40100.2 has been installed!
Verbose mode can be enabled via -v, --verbose option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (3.2.0) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:

framework-arduinoespressif32 3.10006.210326 (1.0.6)
tool-esptoolpy 1.30000.201119 (3.0.0)
tool-mkspiffs 2.230.0 (2.30)
toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 36 compatible libraries
Scanning dependencies...
Dependency Graph
|-- 0.0.1
| |-- 0.4.5+sha.29b8ee4
| |-- 0.6.2
| |-- 2.4.7
| |-- 1.6.0
| |-- 1.0.4
| |-- 4.3.0+sha.6e7e6be
| | |-- 1.0
| |-- 0.0.1+sha.925def4
| |-- 1.0
| |-- 1.0
| | |-- 1.0
Building in release mode
Compiling .pio/build/ESP32/src/main.cpp.o
In file included from /root/.platformio/lib/RFQuack/src/radio/RadioLibWrapper.h:17:0,
from /root/.platformio/lib/RFQuack/src/radio/RFQRFM69.h:5,
from /root/.platformio/lib/RFQuack/src/radio/drivers.h:4,
from src/main.cpp:25:
/root/.platformio/lib/RadioLib/src/RadioLib.h:50:4: warning: #warning "God mode active, I hope it was intentional. Buckle up, lads." [-Wcpp]
#warning "God mode active, I hope it was intentional. Buckle up, lads."
^
In file included from /root/.platformio/lib/RadioLib/src/TypeDef.h:4:0,
from /root/.platformio/lib/RadioLib/src/RadioLib.h:43,
from /root/.platformio/lib/RFQuack/src/radio/RadioLibWrapper.h:17,
from /root/.platformio/lib/RFQuack/src/radio/RFQRFM69.h:5,
from /root/.platformio/lib/RFQuack/src/radio/drivers.h:4,
from src/main.cpp:25:
/root/.platformio/lib/RadioLib/src/BuildOpt.h:124:57: note: #pragma message: RADIOLIB_PLATFORM: ESP32
#define RADIOLIB_PLATFORM "ESP32"
^
/root/.platformio/lib/RadioLib/src/RadioLib.h:55:41: note: in expansion of macro 'RADIOLIB_PLATFORM'
#pragma message "RADIOLIB_PLATFORM: " RADIOLIB_PLATFORM
^
In file included from /root/.platformio/lib/RFQuack/src/radio/drivers.h:4:0,
from src/main.cpp:25:
/root/.platformio/lib/RFQuack/src/radio/RFQRFM69.h: In member function 'virtual float RFQRF69::getRSSI(float&)':
/root/.platformio/lib/RFQuack/src/radio/RFQRFM69.h:140:5: warning: no return statement in function returning non-void [-Wreturn-type]
}
^
Generating partitions .pio/build/ESP32/partitions.bin
[... 127 lines cut ...]
Checking size .pio/build/ESP32/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM: [== ] 16.9% (used 55424 bytes from 327680 bytes)
Flash: [== ] 22.9% (used 300586 bytes from 1310720 bytes)
Building .pio/build/ESP32/firmware.bin
esptool.py v3.0
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Use manually specified: /board
Uploading .pio/build/ESP32/firmware.bin
esptool.py v3.0
Serial port /board
Connecting......................................_____
A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
*** [upload] Error 2
======================================================== [FAILED] Took 155.77 seconds ========================================================

Environment Status Duration

ESP32 FAILED 00:02:35.772
==================================================== 1 failed, 0 succeeded in 00:02:35.772 ====================================================
Makefile:56: recipe for target 'flash' failed
make: *** [flash] Error 1

RChadwick7, 2021-05-29:
I did see both LED's flash, so it looked like it was trying to do something.

joelsernamoreno, 2021-05-29:
That's strange, the firmwares from the Evil Crow RF repository can you upload them or do you also get error?

RChadwick7, 2021-05-29:
Here's the error I get when trying to upload Evil Crow firmware:

Sketch uses 791046 bytes (60%) of program storage space. Maximum is 1310720 bytes.
Global variables use 40352 bytes (12%) of dynamic memory, leaving 287328 bytes for local variables. Maximum is 327680 bytes.
An error occurred while uploading the sketch
esptool.py v3.0-dev
Serial port /dev/ttyUSB0
Traceback (most recent call last):
  File "/home/charles/.arduino15/packages/esp32/tools/esptool_py/3.0.0/esptool.py", line 3682, in <module>
    _main()
  File "/home/charles/.arduino15/packages/esp32/tools/esptool_py/3.0.0/esptool.py", line 3675, in _main
    main()
  File "/home/charles/.arduino15/packages/esp32/tools/esptool_py/3.0.0/esptool.py", line 3329, in main
    esp = chip_class(each_port, initial_baud, args.trace)
  File "/home/charles/.arduino15/packages/esp32/tools/esptool_py/3.0.0/esptool.py", line 263, in __init__
    self._port = serial.serial_for_url(port)
  File "/usr/local/lib/python2.7/dist-packages/serial/__init__.py", line 90, in serial_for_url
    instance.open()
  File "/usr/local/lib/python2.7/dist-packages/serial/serialposix.py", line 325, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 13] could not open port /dev/ttyUSB0: [Errno 13] Permission denied: '/dev/ttyUSB0'

RChadwick7, 2021-05-29:
Also, I don't know if it makes a difference, but I'm running Ubuntu 16.04 in a VM under Windows.

joelsernamoreno, 2021-05-29:
On virtual machines it does not work correctly, try to do it on a physical machine.

Anyway, I'm seeing that the error you are getting is that you don't have permissions on /dev/ttyUSB0

If this still does not work you can contact me on twitter (@JoelSernaMoreno) and I can connect remotely to your computer to fix this

RChadwick7, 2021-05-29:
Unfortunately, I don't have a Linux machine. Do you know if RFQuack could be flashed in Windows?

joelsernamoreno, 2021-05-30:
We tried to flash rfquack on windows but we didn't succeed

You can try it or ask in the rfquack repository

joelsernamoreno, 2021-05-30:
First try flashing the Evil Crow RF firmware with your physical windows. With this we can know that your device does not have any factory problem

RChadwick7, 2021-05-30:
I successfully flashed using Arduino under Windows. Looks like you were right, the issue is with the VM. Too bad RFQuack doesn't have an .INO sketch.

joelsernamoreno, 2021-05-30:
You can try to flash rfquack with Windows. If this doesn't work you will have to open issue in the rfquack repository or flash this with linux

joelsernamoreno, 2021-06-01:
I close this issue. If you have any problems, open another issue or contact me directly on twitter

Thanks

closed 2021-06-01 by joelsernamoreno