joelsernamoreno/EvilCrow-RF #11: LED programming

Azanot, 2021-07-04:
Hi Joel, congratulations for your work.
I bought 2 devices.
Why not use the 3 LEDs to have a better understanding of the operation of the device?! 
Also use a led to indicate signal reception as a "tester"
Example:
LED 1 = Power on signal (As now) :)
LED 2 = On when receiving signal / Flashing if signal in log memory.
LED 3 = On steady or flashing when transmitting a signal.
Thanks for your attention.

joelsernamoreno, 2021-07-04:
Hi!

Yes, you can turn the LEDs on and off with a task. This has not been implemented because each person has to decide when to turn on and off the leds.

The pins are as follows:

led_1 io12
led_2 io13
led_3 io15

Thanks!

Azanot, 2021-07-04:
Thank you for your reply. Unfortunately I don't have the skills to program. Your devices fascinate expert and non-expert users. 
For this reason i thought you would program this and other function as an example "De Bruijn attack", "ROLLJAM" ecc..
Expert user will change the code, non-expert user (like me) will not have the function. Thanks for your attention. My is only suggestion that can increase your sales ;)