joelsernamoreno/EvilCrow-RF #13: Cannot connect to device using RFQuack

Pr0l1f1c, 2021-07-23:
I've been having the same issue since I received my EvilCrowRF-RF devices. I keep getting "Wrong dongle id" no matter what I insert as q.dongle(id).

When attempting to adjust any settings using "q.radioA" or "q.radioB" I receive... "'RFQuack' object has no attribute 'radioA'.

As of now, outside of using your available sketches, it appears that I'm officially stuck at using RFQuack in conjunction with EvilCrowRF-RF. 

I teach a class on various radio topics and I wish to purchase more devices in order to use RFQuack in a much more compact and simpler form instead of having my class attendees build they're own RFQuack devices (the headaches that come along with this are absurd lol).

joelsernamoreno, 2021-07-23:
Hi!

I think this is not an Evil Crow RF issue, it seems that the RFQuack firmware is currently not working.

Check my issue:

https://github.com/rfquack/RFQuack-cli/issues/7

https://github.com/rfquack/RFQuack/issues/9

Maybe it is possible to use an older version of rfquack, I will try this when I have free time.

Thanks!

joelsernamoreno, 2021-07-25:
Hi!

Try this:

[rfquack-old.pdf](https://github.com/joelsernamoreno/EvilCrow-RF/files/6873232/rfquack-old.pdf)

closed 2021-07-30 by joelsernamoreno