joelsernamoreno/EvilCrow-RF #14: When new stock is available? (asked 09/2021)

bosb, 2021-08-29:
Can't find an expectation when new devices are available....so how many to expect and when?

In reference to:
I close this issue. If you ever need to know when new stock is available, please open a new issue

Thanks

_Originally posted by @joelsernamoreno in https://github.com/joelsernamoreno/EvilCrow-RF/issues/8#issuecomment-852558753_

joelsernamoreno, 2021-08-29:
Hi!

The cost of electronic components is very high now, the manufacturer has decided to stop manufacturing Evil Crow RF until prices stabilize

I will try to advise here when the manufacturer gets back to work with this

Thanks!

joelsernamoreno, 2021-09-30:
Stock now

https://m.es.aliexpress.com/item/1005002476481892.html

closed 2021-10-04 by joelsernamoreno