joelsernamoreno/EvilCrow-RF #18: Problems with TX

skaken, 2021-11-17:
I’ve got my hands on a second batch unit and I’m playing around with it but I can’t seem to get it transmitting. I flashed it with the newer RAWv2.2-NewerInterface and everything seems to work on the RX side of things. My test object is a wireless doorbell at 433.92mhz, which I believe uses a simple ASK/OOK but I also tried FSK. Sniffing seems to work and I got everything in the logs, but transmitting either raw data or binary data doesn’t work and I get no errors. Do you have any idea if I’m doing something stupidly wrong or have missed something very obvious? (Should any of the LEDs indicate something during RX/TX?)  I’m a little unsure if I’m using the RxBW, Deviation and Data rate correct, and I would really appreciate some insight into this. 
Thank you very much for all your work, Joel!

joelsernamoreno, 2021-11-17:
Hello!

Can you check with an SDR that the device is transmitting?

Can you join the discord group to be able to analyze the signal?

Thanks

skaken, 2021-11-17:
Thanks for your quick response, Joel! I haven’t got as far as getting myself an SDR yet(hard to pick one nowadays), but I might have to get myself a cheap RTL-SDR for the minor things.

I’ll do a bit more testing on some other RF equipment if you don’t have something obvious I should check, and join the discord group!

joelsernamoreno, 2021-11-17:
If this is ASK this should work:

Freq: 433.92
Mod: ASK/OOK
RxBw: between 58 and 100
Datarate: 5

Then go to log and copy the raw data, if this does not work you can try with the corrected raw data

Go to RAW TX and transmit the data by adding a comma at the end

You can join the discord channel to help you better if this doesn't work

I close this issue

Thanks

closed 2021-11-17 by joelsernamoreno