joelsernamoreno/EvilCrow-RF #21: Adding oled to ecrf?

realone414, 2022-02-07:
How hard would it be to add an oled screed to the board using an arduino oled and header pins

joelsernamoreno, 2022-02-07:
There are currently no visible pads to solder this. You will need to solder this directly to the pins of the ESP

realone414, 2022-02-08:
Is there a license I can buy from you to purchase the complete schematics,
bom and gerber files to make only 1? I would like to build 1 and try to add
a couple functions as well as redesign it a little and make extremely good
quality enclosures with my creality cr-30 3d printer. I would be more than
happy to share all of my work back with you and send you anything you need.
Please help me, I have been completely obsessed with this project and can't
sleep because of all the ideas I have.  I can't find a single one to buy
anywhere and I'm so impatient.  Thank you for the reply, I how to hear from
you soon.

closed 2022-03-03 by joelsernamoreno