joelsernamoreno/EvilCrow-RF #25: Where are the instructions?

WULFFJ, 2022-12-24:
I bought three of these to do some testing on.  I am newer to RF, but not programming.  Everything installed fine.  However, the lack of instructions is horrible.

What is the power jammer parameter for or what should be assigned?

Do all the other functions of this device call for another sketch?  This device gets advertised as an all in one???

Where are any guides on anything 2.4ghz related for the v2?  

How does this integrate with URH?  You have a video of it, but its selecting another device.  Is it another device that is listening to the Evil Crow?

Overall, these directions don't give a good direction of where to start and is rather annoying.

Commander-Crash, 2022-12-26:
Agree. The web interface can use some explaining. I wish it was compatible with RFCAT but it seems like it is not a good alt for the YS1

Crsarmv7l, 2023-04-09:
> Agree. The web interface can use some explaining. I wish it was compatible with RFCAT but it seems like it is not a good alt for the YS1

I have been looking for something like rfcat but with less plug/unplug and more stable. Thought it might be rfquack but that has its own issues.

closed 2026-01-28 by joelsernamoreno