joelsernamoreno/EvilCrow-RF #29: Getting error when trying to test compile in arduino IDE

kavemang, 2024-07-08:
`
In file included from /home/kaveman/.arduino15/packages/esp32/hardware/esp32/3.0.2/libraries/WebServer/src/HTTP_Method.h:4,
                 from /home/kaveman/.arduino15/packages/esp32/hardware/esp32/3.0.2/libraries/WebServer/src/WebServer.h:30,
                 from /home/kaveman/Arduino/libraries/ElegantOTA/src/ElegantOTA.h:73,
                 from /home/kaveman/Arduino/libraries/EvilCrow-RF/EvilCrowRF-RAWv2/RAWv3.0-NewInterface/RAWv3.0-NewInterface.ino:6:
/home/kaveman/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32/include/http_parser/http_parser.h:95:6: error: 'HTTP_DELETE' conflicts with a previous declaration
   95 |   XX(0,  DELETE,      DELETE)       \
      |      ^
/home/kaveman/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32/include/http_parser/http_parser.h:138:45: note: in definition of macro 'XX'
  138 | #define XX(num, name, string) HTTP_##name = num,
      |                                             ^~~
/home/kaveman/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32/include/http_parser/http_parser.h:139:3: note: in expansion of macro 'HTTP_METHOD_MAP'
  139 |   HTTP_METHOD_MAP(XX)
      |   ^~~~~~~~~~~~~~~
In file included from /home/kaveman/Arduino/libraries/EvilCrow-RF/EvilCrowRF-RAWv2/RAWv3.0-NewInterface/RAWv3.0-NewInterface.ino:5:
/home/kaveman/Arduino/libraries/ESPAsyncWebServer/src/ESPAsyncWebServer.h:64:3: note: previous declaration 'WebRequestMethod HTTP_DELETE'
   64 |   HTTP_DELETE  = 0b00000100,
      |   ^~~~~~~~~~~
/home/kaveman/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32/include/http_parser/http_parser.h:96:6: error: 'HTTP_GET' conflicts with a previous declaration
   96 |   XX(1,  GET,         GET)          \
      |      ^
/home/kaveman/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32/include/http_parser/http_parser.h:138:45: note: in definition of macro 'XX'
  138 | #define XX(num, name, string) HTTP_##name = num,
      |                                             ^~~
/home/kaveman/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32/include/http_parser/http_parser.h:139:3: note: in expansion of macro 'HTTP_METHOD_MAP'
  139 |   HTTP_METHOD_MAP(XX)
      |   ^~~~~~~~~~~~~~~
/home/kaveman/Arduino/libraries/ESPAsyncWebServer/src/ESPAsyncWebServer.h:62:3: note: previous declaration 'WebRequestMethod HTTP_GET'
   62 |   HTTP_GET     = 0b00000001,
      |   ^~~~~~~~
/home/kaveman/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32/include/http_parser/http_parser.h:97:6: error: 'HTTP_HEAD' conflicts with a previous declaration
   97 |   XX(2,  HEAD,        HEAD)         \
      |      ^
/home/kaveman/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32/include/http_parser/http_parser.h:138:45: note: in definition of macro 'XX'
  138 | #define XX(num, name, string) HTTP_##name = num,
      |                                             ^~~
/home/kaveman/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32/include/http_parser/http_parser.h:139:3: note: in expansion of macro 'HTTP_METHOD_MAP'
  139 |   HTTP_METHOD_MAP(XX)
      |   ^~~~~~~~~~~~~~~
/home/kaveman/Arduino/libraries/ESPAsyncWebServer/src/ESPAsyncWebServer.h:67:3: note: previous declaration 'WebRequestMethod HTTP_HEAD'
   67 |   HTTP_HEAD    = 0b00100000,
      |   ^~~~~~~~~
/home/kaveman/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32/include/http_parser/http_parser.h:98:6: error: 'HTTP_POST' conflicts with a previous declaration
   98 |   XX(3,  POST,        POST)         \
      |      ^
/home/kaveman/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32/include/http_parser/http_parser.h:138:45: note: in definition of macro 'XX'
  138 | #define XX(num, name, string) HTTP_##name = num,
      |                                             ^~~
/home/kaveman/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32/include/http_parser/http_parser.h:139:3: note: in expansion of macro 'HTTP_METHOD_MAP'
  139 |   HTTP_METHOD_MAP(XX)
      |   ^~~~~~~~~~~~~~~
/home/kaveman/Arduino/libraries/ESPAsyncWebServer/src/ESPAsyncWebServer.h:63:3: note: previous declaration 'WebRequestMethod HTTP_POST'
   63 |   HTTP_POST    = 0b00000010,
      |   ^~~~~~~~~
/home/kaveman/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32/include/http_parser/http_parser.h:99:6: error: 'HTTP_PUT' conflicts with a previous declaration
   99 |   XX(4,  PUT,         PUT)          \
      |      ^
/home/kaveman/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32/include/http_parser/http_parser.h:138:45: note: in definition of macro 'XX'
  138 | #define XX(num, name, string) HTTP_##name = num,
      |                                             ^~~
/home/kaveman/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32/include/http_parser/http_parser.h:139:3: note: in expansion of macro 'HTTP_METHOD_MAP'
  139 |   HTTP_METHOD_MAP(XX)
      |   ^~~~~~~~~~~~~~~
/home/kaveman/Arduino/libraries/ESPAsyncWebServer/src/ESPAsyncWebServer.h:65:3: note: previous declaration 'WebRequestMethod HTTP_PUT'
   65 |   HTTP_PUT     = 0b00001000,
      |   ^~~~~~~~
/home/kaveman/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32/include/http_parser/http_parser.h:102:6: error: 'HTTP_OPTIONS' conflicts with a previous declaration
  102 |   XX(6,  OPTIONS,     OPTIONS)      \
      |      ^
/home/kaveman/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32/include/http_parser/http_parser.h:138:45: note: in definition of macro 'XX'
  138 | #define XX(num, name, string) HTTP_##name = num,
      |                                             ^~~
/home/kaveman/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32/include/http_parser/http_parser.h:139:3: note: in expansion of macro 'HTTP_METHOD_MAP'
  139 |   HTTP_METHOD_MAP(XX)
      |   ^~~~~~~~~~~~~~~
/home/kaveman/Arduino/libraries/ESPAsyncWebServer/src/ESPAsyncWebServer.h:68:3: note: previous declaration 'WebRequestMethod HTTP_OPTIONS'
   68 |   HTTP_OPTIONS = 0b01000000,
      |   ^~~~~~~~~~~~
/home/kaveman/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32/include/http_parser/http_parser.h:128:6: error: 'HTTP_PATCH' conflicts with a previous declaration
  128 |   XX(28, PATCH,       PATCH)        \
      |      ^~
/home/kaveman/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32/include/http_parser/http_parser.h:138:45: note: in definition of macro 'XX'
  138 | #define XX(num, name, string) HTTP_##name = num,
      |                                             ^~~
/home/kaveman/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32/include/http_parser/http_parser.h:139:3: note: in expansion of macro 'HTTP_METHOD_MAP'
  139 |   HTTP_METHOD_MAP(XX)
      |   ^~~~~~~~~~~~~~~
/home/kaveman/Arduino/libraries/ESPAsyncWebServer/src/ESPAsyncWebServer.h:66:3: note: previous declaration 'WebRequestMethod HTTP_PATCH'
   66 |   HTTP_PATCH   = 0b00010000,
      |   ^~~~~~~~~~
/home/kaveman/Arduino/libraries/EvilCrow-RF/EvilCrowRF-RAWv2/RAWv3.0-NewInterface/RAWv3.0-NewInterface.ino: In function 'void force_reset()':
/home/kaveman/Arduino/libraries/EvilCrow-RF/EvilCrowRF-RAWv2/RAWv3.0-NewInterface/RAWv3.0-NewInterface.ino:280:3: error: 'esp_task_wdt_add' was not declared in this scope
  280 |   esp_task_wdt_add(NULL);
      |   ^~~~~~~~~~~~~~~~
/home/kaveman/Arduino/libraries/EvilCrow-RF/EvilCrowRF-RAWv2/RAWv3.0-NewInterface/RAWv3.0-NewInterface.ino: In function 'void setup()':
/home/kaveman/Arduino/libraries/EvilCrow-RF/EvilCrowRF-RAWv2/RAWv3.0-NewInterface/RAWv3.0-NewInterface.ino:846:3: error: 'AsyncElegantOTA' was not declared in this scope; did you mean 'ElegantOTA'?
  846 |   AsyncElegantOTA.begin(&controlserver);
      |   ^~~~~~~~~~~~~~~
      |   ElegantOTA

exit status 1

Compilation error: 'esp_task_wdt_add' was not declared in this scope
`

kavemang, 2024-07-09:
wrong place

joelsernamoreno, 2026-01-28:
Hello

I’ve released a new firmware update.
I’m going to close all existing issues. If you encounter any problems with the new firmware, or if any previous issues haven’t been resolved, please open a new issue.

Thank you!