joelsernamoreno/EvilCrow-RF #30: SPIFFS depreciated

mattgp, 2025-02-02:
Hi,

The code won't compile in arduino IDE due to the above.

Is there a workaround, went to discord for support but the link has expired.

BR

Matt.

joelsernamoreno, 2025-02-14:
Hello

Yes, the ECRF V1 and V2 firmware are broken due to the latest library updates, I am programming the firmware from scratch again

Unfortunately this is the Evil Crow RF V1 firmware and has no alternative firmware, only Evil Crow RF V2

I will respond on this issue when I finish programming the firmware

Thanks

mattgp, 2025-02-15:
Hey,Thanks for the update and all the hard work.I actually worked around the issue by using an old library from another unrelated GitHub project, this created 2 minor compile errors which I just commented out.The result was firmware that worked well enough to log in and use OTA to install RAT firmware which I’ll use for the time being.The only nuance I noted was if I reboot the ECRF the firmware revered to my semi broken version ie non persistent uodste - weird.I fixed this by updating the rat firmware to itself using OTA a second time before rebooting again which fixed it.All the best,Matt.